dvmarinoff/Auuki #213: No cadence reading from Wahoo cadence sensor

opentree27, 2024-11-19:
I've just picked up a wahoo cadence sensor and after connecting to it I get no readout for cadence data. Not even a 0 but a -- instead. I've also tried connecting via BLE but it does not connect.

Here the link to the sensor on wahoos website:
https://uk.wahoofitness.com/devices/bike-sensors/wahoo-rpm-cadence-sensor

opentree27, 2024-11-20:
I have tried both the latest and stable versions and get the same result

dvmarinoff, 2024-11-20:
Just to confirm, do you have the device settings correct?

By default Controllable is set to read cadence, the label cadence is white when selected and gray when deselected.

In the first image it's set to Controllable:
[image]

In the second image to Speed and Cadence sensor:
[image]

opentree27, 2024-11-20:
That's fixed it thanks. Not to make excuses for myself, but it is not very obvious that this is needed. I assumed that if you connected to a dedicated cadence sensor that that would always be preferred and also if there is no cadence from the controllable then any alternate cadence data available would be used. Appreciate this is in development but, maybe something for the future :)

dvmarinoff, 2024-11-20:
Totally, agree it's just that there are way too many options. It's a lot of complexity. I explored the automatically set the specialized device to read the value road, but it leads to even more complexity, considering that some HR monitors have cadence (but only indoors ?!), and so on. I am going to stay with the current approach where the Controllable and Power meter are default first choices and everything else needs to be configured by the User.

I just need to make that more obvious, maybe more visible labels or switches (iOS style).

closed 2024-11-20 by dvmarinoff