dvmarinoff/Auuki #214: improve device settings

dvmarinoff, 2024-11-21:
The device settings page doesn't make it clear that the User needs to manually toggle on the data channels (power, cadence heart rate) on some classes of devices. @opentree27, @jimdavcon My current idea is to add toggles, will something of the sort help?

Here are two toggle styles. I am more inclined to go with the iOS style, it's more generally recognizable as a toggle, but the Chrome one is maybe a bit more inline with the overall design of this app (which is inspired by the Google Podcasts app).

iOS style toggle:
 
[image]

Google Chrome style toggle:
[image]

jimdavcon, 2024-11-21:
2 suggestions:

Which ever style you choose, make the BT thing on the left toggle the same
way.

Make all these toggles use the same colors: red/green, white/blue..  one or
the other.

I don't have a strong opinion about the toggle style itself although, since
I mostly use the app on a cell phone, bigger buttons are preferred.

Thanks,
Jim

dvmarinoff, 2024-11-21:
Ok, one more toggle design.

Fewer colors, 'flat style', just white and gray as the rest of the buttons (well except for the bluetooth connection indicators).
And the whole area will be clickable (text label + text value + toggle button).

[image]

jimdavcon, 2024-11-21:
> Or more inline with @jimdavcon <https://github.com/jimdavcon> suggestion,
> something like that. But I am not sure if this is not too much color for me.
> Screenshot.2024-11-21.at.18.27.46.png (view on web)

Would be perfect if only the circles were colored but looks pretty good to
me as is.

-Jim

dvmarinoff, 2024-11-21:
Thanks!

 will push to dev the gray and white version for now, and will try to play with more color later ...

effectpears, 2024-11-23:
Personally, I like the colored version (green/red) better, but every switch is an improvement.

opentree27, 2024-11-23:
Personally, I agree that the chrome style looks the best. I feel like the green and red toggles have way too much color and causes it to lose its meaning a bit. Particularly if you aren't using things like MOXY and Cadence sensors the colours really do not make sense. I strongly feel that if you do not have the device connected there is no reason for the toggles to have any colour or imply any ability to be active unless the device is actually connected.

dvmarinoff, 2024-11-28:
Thanks all for the feedback!

the feature is now live on both master and develop branches. Closing the issue.

jimdavcon, 2024-11-30:
Tried these updates today but I'm not sure it works like I expected. I can
turn on BLE power meter and I can turn on power, cadence and speed services
but I can't switch them off once they're on.

The power meter cadence turns off if I select cadence from the speed and
cadence BLE features but I can't turn off the services in power meter
directly.

Maybe that's how you meant it to work but it seems a bit intuitive that
these on/off switches don't work both ways themselves.

- Jim

dvmarinoff, 2024-11-30:
If there are multiple devices connected and they have overlapping data, say the controllable has cadence, but also power meter has cadence, a dedicated cadence sensor has cadence, and some heart rate monitors also have cadence.

In that case the application needs to figure out which one of those cadences to display and to record in the .FIT file (the data export format that goes into strava, garmin etc). Currently there can be only one primary cadence at a time, so this interface acts as switch case. Same goes for the rest of the data channels, power, heart rate, and so on.

I am also assuming that most users will use the same setup, each time they ride so the setup is being preserved until the user chooses to change it. Unlike Zwift which forgets my setup and tries to guess my intensions, usually wrong, and than often overwrites my device configuration in the background, while I am riding.

My idea behind this type of design is that the user needs to configure those settings
1. when they add a new device to their setup,
2. when some device is acting out during the ride session

So the setup gets saved for future reuse and is also quickly available for change at a one click distance. No need to open up a series of popups and nested menu items, when you want to manually switch to a different device during ride (say I forgot to charge my power meter pedals or the cadence sensor has difficulties with interference, or just bad data on a given day happens even to the best heart rate monitors). I usually use 2 HRMs Apple watch and a Polar strap.

I am open to suggestions to make this better and more intuitive given the listed requirements.

jimdavcon, 2024-11-30:
I understand now...   So these new controls actually behave like radio
buttons not as toggle switches. One of each measurement is always 'on' even
if I don't want to record it. I guess I didn't understand that from our
original chat so I thought I could just turn off any measurement that I
don't have or don't want to record/share.

In my case, my power meter has valid power and cadence measurements but my
speed is not correct so I tried to turn off it's switch but couldn't.
Instead, I suppose I can turn on speed from a device I don't have so that
it's always zero?

So the requirements are something like:

1. Allow the user to select a single source of each measurement.
2. Make user selections persistent

I'm suggesting one additional requirement:

3. Allow the user to disable measurements that are undesired.

This added capability would make a better case for using colors
(enabled=green, enabled but conflict=yellow, disabled=gray?) but I totally
acknowledge that the logic to keep track of which measurements are enabled
or disabled gets very messy if the same measurement from multiple devices
can be enabled at the same time.

Now that I understand your intent, I realize that things are more
complicated than I understood previously. Please disregard these
suggestions if they make things too complicated!

And, by the way, THANK YOU! Flux (Auuki?) continues to work splendidly.

- Jim

dvmarinoff, 2024-11-30:
Hey, thanks for the feedback, it’s really important to know if a feature is headed in the right direction or not. It’s just that this particular feature is maybe one of the top 3 on difficulty. It has to deal with a lot of complexity on hardware level and the personal preferences of different potential users while maintaining a clean and simple UI at a minimal click distance. I don’t think any of the existing apps out there has a good solution, so I hope we can figure out at least something different for Auuki.

> 3. Allow the user to disable measurements that are undesired.

That’s a valid point. Although I’d need to run some checks to see if it is viable. My worry is that strava and garmin would look with suspicion on activities that don’t have speed and distance.

> but my speed is not correct

Just to note that in settings you can find a switch named Simulation. It will switch between ‘speed’ where the speed is recorded as coming from the trainer or other speed enabled device, and the other option is ‘power’ where the app will use a physics model and will calculate the speed based on the power input, slope and other simulation parameters.