dvmarinoff/Auuki #216: Keyboard Shortcuts

effectpears, 2024-12-01:
Who hasn't experienced it? You're struggling with the hard interval and wish it would end. Your eyes are full of sweat and the keys are all so small - how about putting the lap button on the space bar and pause on P. Since there's the new option for auto-pause.

Or there's an option to customize it, but that seems complicated to me.

dvmarinoff, 2024-12-01:
We already have some keyboard shortcuts. They are documented here: https://github.com/dvmarinoff/Auuki/wiki/Keyboard-Shortcuts.

- `L` key is for Lap/Skip button
- `Spacebar` is for start/resume toggle

You can also change control modes and targets in increments of 10 watts for power, 10 for resistance and 0.5% for slope, with the up and down arrow keys.

But thanks for the report! I took a closer look at the functionality and found a bug with Spacebar not starting the workout only the watch. A fix is ready, will publish in a few hours.

effectpears, 2024-12-01:
Thank you for your detailed answer. I still need to learn how to participate properly and clearly state what I want.
In practice, you will need to skip more often and so it is practical if it is the most visible/accessible key.

I suggest swapping the shortcuts for pause and skip.

dvmarinoff, 2024-12-01:
I see now ... it makes sense.

one issue would be how to communicate the change for people already used to the Spacebar toggling pause/resume
Will think about it.