dvmarinoff/Auuki #217: Strava upload

Rw721, 2024-12-08:
Apologies if I'm being a bit stupid here as I'm not very good at this kind of stuff but I see in a recent update the upload to Garmin link has been replaced with what looks like a Strava upload. 
Unfortunatley I don't seem to be able to get it to work on my laptop . There is an error message saying:  "No internet connection or the API service is currently offline"
I've attached a file below
[Auuki Strava upload.docx](https://github.com/user-attachments/files/18053616/Auuki.Strava.upload.docx)

dvmarinoff, 2024-12-08:
Yes, sorry about that, this feature is not yet finished. The front-end UI that you are seeing is all done and ready, but I still have some work to do in order to secure the back-end service that will communicate with Strava and other platforms. This being a free to use project may attract malicious actors (bots) and I need to have some extra security steps on the back-end to prevent abuse.

Expect updates soon. Thanks for the Patience!

dvmarinoff, 2024-12-08:
Here are some screenshots to show how the new feature will look.

You'd need to register a profile with email and password. Connect one of the supported services, which will be Strava, Intervals.icu and Training Peaks at the start (Garmin will remain as before just a link to the import page).

Now you will be able to upload a finished workouts to a service from the Activities List.

Also you'd be able to pull daily planned workouts from Intervals.icu and Training Peaks at the Workouts tab.

<table>
<tr>
<td>
[image]
</td>
<td>
[image]
</td>
<td>
[image]
</td>
</tr>
</table>

Rw721, 2024-12-09:
> <https://github.com/user-attachments/assets/4c8a3008-feec-4c69-af36-da5f86b35e51> Screenshot.2024-12-08.at.22.14.51.png
> <https://github.com/user-attachments/assets/e6035d14-f068-4b75-988d-134a6ee95c98> Screenshot.2024-12-08.at.22.15.38.png

Ah brilliant, that is going to be so good.

Many thanks

dvmarinoff, 2024-12-22:
A little update the feature is working solid. The last step is to go through the Strava review process and get larger user count limit. Currently the app has a user limit of 1. 

Meanwhile the Intervals.icu integration is ready for preview you can check it out on https://dev.auuki.com

It works on my machine, but it might have some issues with the UI for registering and login.

A page reload should be able to resolve most issues.

karlacio, 2024-12-22:
Hi, I'm trying to use it with intervals.

I can see buttons to download planned activities and to upload but no way to login to the service. 

If I press the upload or download buttons for intervals I get a popup "Your session is over. You need to login again."

I tried to reload page without success. 

I'm using it on Android as app.

dvmarinoff, 2024-12-22:
Hey @karlacio,

Thanks for the feedback! I see I have configured the wrong redirect-url for the app. It's fixed now.
When you go to the settings tab you should be able to see the login/register UI. Here is how the procedure goes:

<table>
<tr>
<td>[image]</td>
<td>[image]</td>
<td>[image]</td>
</tr>
<tr>
<td>[image]</td>
<td>[image]</td>
<td>[image]</td>
</tr>
<tr>
<td>[image]</td>
</tr>
</table>

karlacio, 2024-12-22:
Hi Dimitar,

Just tried and it works (with intervals).
Maybe tomorrow I'll do a "planned" workout and I'll upload data to intervals.

I want to report a small problem I had:
after registering I proceeded with intervals connection. After entering the email address and OTP I got an error page (from intervals):

> org.springframework.web.method.annotation.MethodArgumentTypeMismatchException: Failed to convert value of type 'java.lang.String' to required type 'int'; nested exception is java.lang.NumberFormatException: For input string: "undefined"

going back and then using auuki the connection seems to be working.

Thanks again for your work!

dvmarinoff, 2024-12-22:
Ah I just found why this is happening ... the fix will require a subtle redesign of the way the feature is being configured.
Will fix it soon. But meanwhile you will need to reload the page after registering and login for the first time. It's just that first login after register thats causing the issue.

Thanks for the report!

Rw721, 2024-12-22:
Just to check, I'm I supposed to be able to see the Strava upload in version 0.1.69?
All I see at the moment is what was there previously:

[image]

dvmarinoff, 2024-12-23:
@Rw721 

currently we have only a preview of the Register/Login system and the Interval.icu integration on the `dev.auuki.com` domain. I am waiting on Strava for review of my app so it is not yet available. It should be version `0.1.76`.

Also the first login issues should be fixed now.

karlacio, 2024-12-23:
Today I did a download planned workout and then upload of data from/to intervals and it worked perfectly!

napfbike, 2024-12-24:
Hi together

What an great christmas gift 😉

I could connect to intervals.icu and received my todays training from the calendar. The finished training has been uploaded flawlessly.

Thank you so much @dvmarinoff for this great app!

It keeps me focused on the workout, the GUI is just perfect.

In this sense: "Less is more!"

Best regards and enjoy your holidays 🎄
Pius

dvmarinoff, 2024-12-27:
Update on Strava, the API submission was just approved so I guess this can go live as early as tomorrow.

dvmarinoff, 2024-12-27:
Alright Strava is also currently available for testing on [dev.auuki.com](https://dev.auuki.com)

Rw721, 2024-12-27:
Looking good so far i.e, I've managed to register.
Just need to do a workout now!

dvmarinoff, 2024-12-28:
Just pushed both Strava integration and Intervals.icu integration to production. You'd able to use both on [auuki.com](https://auuki.com) and [dev.auuki.com](https://dev.auuki.com).

- Strava should be all done.
- Intervals.icu is working, but once we get some more feedback I'll write David from Intervals to turn on the Intervals.icu UI for Auuki and I'll drop an announcement in the forums, but first let's get some feedback and confirm everything is working.

 I'd appreciate some feedback on both integrations. Thanks for the help everyone!

closed 2024-12-30 by dvmarinoff

napfbike, 2024-12-31:
Hi
Just some feedback.

I synchronized one TrainerDay workout over Intervals.icu to Auuki.

The upload to Strava of the finished workout was perfect.

[image]

I would be great if the name of the workout would be added in Strava. And a graph with the laps. I have added the screenshot from Intervals to Strava.

However adding these extra information might be paid features or allowed only  to the big boys.

dvmarinoff, 2024-12-31:
@napfbike 

Nice! Thanks for the feedback! 

> would be great if the name of the workout would be added in Strava

sure will add this in the next iteration

> And a graph with the laps

Last time I checked uploading images to Strava was limited to only a few select applications. Will check again, but an svg image can be created from the already recorded data easily so maybe some kind of proxy solution might work.

> I synchronized one TrainerDay workout over Intervals.icu to Auuki.

Would you benefit from a TrainerDay integration? Or Intervals.icu is working better as a proxy?

napfbike, 2025-01-01:
@dvmarinoff thank you!

As for the workout name and graph, it would be great to have that implemented in the future.

I can definitely see the point of connecting TrainerDay. 

TD's training interface is quite similar to Auuki's, but in my opinion a bit overloaded. What is striking, however, is that it is only available as an Android or iOS app. You can display the front end on a large screen via a link on a Windows PC, but you can't control anything. This is where I see Auuki as a real help, as everything would run on Windows.

Rw721, 2025-01-01:
Yes, importing the workout name (or offering a dialogue box to give it a
name in case you dont quite do the workout as planned)  prior to uploading
it to Strava would be great'. This would avoid the Strava default of
'afternoon ride' etc and having to go in the rename the workout.

Thanks

dvmarinoff, 2025-01-01:
Just pushed an update with version 0.3.1.

The workout name is added to the strava upload, and the activity is tagged as VirtualRide as well.

Just did half of Chilli Pepper +1 for an end-to-end test:

[image]

napfbike, 2025-01-03:
Hey!

Just finished another TrainerDay workout over Intervals.icu
The upload to Strava looks great!

[image]

Thanks!