dvmarinoff/Auuki #220: Accidentally skipping a workout block

Rw721, 2024-12-11:
Hi, sorry, one more question from me.
Sometimes I accidentally skip a workout block and the only way I'm only  able to go back is by re-loading the workout. However  (I think in order not to loose a workout) there is a feature that means it just picks right back where I left off in the workout. 
The only workaround I have found is to delete browsing data.
Is there a way of repeating a workout block without having to resort to this?

Many thanks

dvmarinoff, 2024-12-11:
Not at the moment, but the easiest thing for now would be to add a skip to previous interval button. Will that do the job?

[image]

dvmarinoff, 2024-12-11:
One ontological question: What should the skip back button do with the current interval data wise? Wrap it up as a separate thing before it skips back to the previous, or append it to the previous without creating a new entity?

Rw721, 2024-12-11:
Hmm, good question.
I want to say just append it to the previous.

I wonder if this is why Zwift never implemented this option despite
repeated requests??

Regards