dvmarinoff/Auuki #228: Dock mode behaviour

Lory3881, 2025-01-15:
A question: when you activate dock mode and open the reduced window, can the original window be closed or should it remain active? I could be wrong but I noticed that if I connect the smart trainer on the page and then reduce it to dock, the speed shown would seem to be halved.

dvmarinoff, 2025-01-15:
It's better to close the original window and connect the devices through the Dock Mode window only. Otherwise it may work, but with some issues (unpredictable), it's just the limitations of the technology.

Let me know if that works for you. I haven't checked dock mode in a while, maybe should give it some improvements.

Lory3881, 2025-01-15:
I will sure check and come back to you.

As suggestions, I think it would be useful: 
- whether the dock could be locked in size once opened 
- be able to open auuki directly as a dock without going through the settings page

Dock mode i found to be a winning point for me!

Auuki is really a game changer!

dvmarinoff, 2025-01-16:
> whether the dock could be locked in size once opened

Can you try to describe the particular use case. I am trying to understand the way people are using dock mode.

> be able to open auuki directly as a dock without going through the settings page

I am not sure if this is technically possible, will need to do some research and testing, maybe there is some hack-ish way to achieve a similar result.

Meanwhile I've found some layout issue with dock mode, where the phone landscape mode styles are clashing with the dock mode styles. Will try to fix it:

So the styles that create the balanced 50% / 50% layout here:
[image]

Are doing some damage here, by offsetting the Control Mode Buttons:
[image]

Lory3881, 2025-01-16:
Once the window has been opened in dock mode, the window is always resizable. Resize the window corrupts the viewable fields

dvmarinoff, 2025-01-16:
I see. That's not ideal:

https://github.com/user-attachments/assets/3dbf5536-9ef6-48b9-b831-92943c841180

Will try first to address the `viewable fields getting corrupted` issue. Ideally they should work with any meaningful window size.
If that proves to be not an option will try to restrict the sizing. But the priority is to first support any screen size from Phone to TV.

dvmarinoff, 2025-01-19:
Alright, that was some tough debugging, but Dock mode is getting some responsive improvements.
The issues were caused also by the workout graph refusing to downsize. Now it should behave as expected and fit inside the
container:

https://github.com/user-attachments/assets/79e35c58-c2b2-4604-b154-2f386095d8fc

dvmarinoff, 2025-01-19:
The fixes are currently published only on the development version at dev.auuki.com. Will do a few rides and will push them to the main version if there are no issues.

Lory3881, 2025-01-19:
Good job! I will try it too and give you feedbacks

closed 2025-01-27 by dvmarinoff