dvmarinoff/Auuki #230: video sync

shug0131, 2025-01-21:
Would it be possible to add a feature that allows a video to be synced to a .fit file please.  So the video advances in time according to the distance/timestamp on the fit file. 

A use would be to go and ride your favourite time trial route with a video camera along with bike computer. Then come back to train with it here, during winter evenings, and get the video alongside. 

Golden Cheetah had this. I managed to make it work once or twice. But now with an upgrade to my hardware it fails to connect and seems to be very buggy.

dvmarinoff, 2025-01-21:
I don't know how the whole video to fit file syncing works. And honestly this falls outside the scope of the project. If there is high demand for this kind of feature maybe in the future it will make the list.

Since this project is just vanilla javascript and css, it's not too difficult to be extended. And you can let me know if you need some guidance with the codebase.

cagnulein, 2025-01-30:
If you are interested qz does too since 2 years ago https://github.com/cagnulein/qdomyos-zwift/

dvmarinoff, 2025-01-30:
@cagnulein Should I start posting Rouvy commercials in your projects Issue Tracker? Because If everyone starts playing like this the issue trackers will become unusable.

If you are going to market, I think there are plenty of forums out there which are much more suitable towards conducting this type of activity.

cagnulein, 2025-01-30:
@dvmarinoff i was commenting because @MedTechCD said the incyclist project did this and so qz, and so you can grab the code from qz.
I suggest to measure the words next time, we are both writing open code man!

dvmarinoff, 2025-01-30:
Indeed we do.
Anyway this feature will remain out of scope, so closing the issue.

closed 2025-01-30 by dvmarinoff