dvmarinoff/Auuki #232: fix off error in workout power targets

dvmarinoff, 2025-01-23:
Power targets on Intervals.icu synced workouts seem to be about 2 watts lower.

Workouts from Intervals.icu come encoded as:
```
{
    workout_file_base64: "PD94bWwgdmVyc2lvbj0iMS4wIiBlbmNvZGluZz0iVVRGLTgiIHN0YW5kYWxvbmU9InllcyI/Pgo8d29ya291dF9maWxlPgogICAgPGF1dGhvcj5EaW1pdGFyIE1hcmlub3Y8L2F1dGhvcj4KICAgIDxuYW1lPkt1dXNpIEthbGFhPC9uYW1lPgogICAgPHNwb3J0VHlwZT5iaWtlPC9zcG9ydFR5cGU+CiAgICA8dGFncy8+CiAgICA8d29ya291dD4KICAgICAgICA8U3RlYWR5U3RhdGUgc2hvd19hdmc9IjEiIFBvd2VyPSIwLjQxNyIgRHVyYXRpb249IjYwMCIvPgogICAgICAgIDxTdGVhZHlTdGF0ZSBzaG93X2F2Zz0iMSIgUG93ZXI9IjAuNTc4IiBEdXJhdGlvbj0iNjAwIi8+CiAgICAgICAgPFN0ZWFkeVN0YXRlIHNob3dfYXZnPSIxIiBQb3dlcj0iMC42MTkiIER1cmF0aW9uPSI2MDAiLz4KICAgICAgICA8U3RlYWR5U3RhdGUgc2hvd19hdmc9IjEiIFBvd2VyPSIwLjY2OSIgRHVyYXRpb249IjYwMCIvPgogICAgICAgIDxTdGVhZHlTdGF0ZSBzaG93X2F2Zz0iMSIgUG93ZXI9IjAuNjE5IiBEdXJhdGlvbj0iNjAwIi8+CiAgICAgICAgPFN0ZWFkeVN0YXRlIHNob3dfYXZnPSIxIiBQb3dlcj0iMC42NjkiIER1cmF0aW9uPSI2MDAiLz4KICAgICAgICA8U3RlYWR5U3RhdGUgc2hvd19hdmc9IjEiIFBvd2VyPSIwLjY5OCIgRHVyYXRpb249IjE4MCIvPgogICAgICAgIDxTdGVhZHlTdGF0ZSBzaG93X2F2Zz0iMSIgUG93ZXI9IjAuNTc4IiBEdXJhdGlvbj0iMTIwIi8+CiAgICAgICAgPFN0ZWFkeVN0YXRlIHNob3dfYXZnPSIxIiBQb3dlcj0iMC41IiBEdXJhdGlvbj0iMzAwIi8+CiAgICA8L3dvcmtvdXQ+Cjwvd29ya291dF9maWxlPgo="
}
```

Which decodes to:
```
<?xml version="1.0" encoding="UTF-8" standalone="yes"?>
<workout_file>
    <author>Dimitar Marinov</author>
    <name>Kuusi Kalaa</name>
    <sportType>bike</sportType>
    <tags/>
    <workout>
        <SteadyState show_avg="1" Power="0.417" Duration="600"/>
        <SteadyState show_avg="1" Power="0.578" Duration="600"/>
        <SteadyState show_avg="1" Power="0.619" Duration="600"/>
        <SteadyState show_avg="1" Power="0.669" Duration="600"/>
        <SteadyState show_avg="1" Power="0.619" Duration="600"/>
        <SteadyState show_avg="1" Power="0.669" Duration="600"/>
        <SteadyState show_avg="1" Power="0.698" Duration="180"/>
        <SteadyState show_avg="1" Power="0.578" Duration="120"/>
        <SteadyState show_avg="1" Power="0.5" Duration="300"/>
    </workout>
</workout_file>
```

Then the parser outputs:
```
{
    "meta": {
        "duration": 4200,
        "planned": true
    },
    "intervals": [{
            "duration": 600,
            "steps": [{
                    "show_avg": "1",
                    "power": 0.417,
                    "duration": 600
                }]
        }, {
            "duration": 600,
            "steps": [{
                    "show_avg": "1",
                    "power": 0.578,
                    "duration": 600
                }]
        }, {
            "duration": 600,
            "steps": [{
                    "show_avg": "1",
                    "power": 0.619,
                    "duration": 600
                }]
        }, {
            "duration": 600,
            "steps": [{
                    "show_avg": "1",
                    "power": 0.669,
                    "duration": 600
                }]
        }, {
            "duration": 600,
            "steps": [{
                    "show_avg": "1",
                    "power": 0.619,
                    "duration": 600
                }]
        }, {
            "duration": 600,
            "steps": [{
                    "show_avg": "1",
                    "power": 0.669,
                    "duration": 600
                }]
        }, {
            "duration": 180,
            "steps": [{
                    "show_avg": "1",
                    "power": 0.698,
                    "duration": 180
                }]
        }, {
            "duration": 120,
            "steps": [{
                    "show_avg": "1",
                    "power": 0.578,
                    "duration": 120
                }]
        }, {
            "duration": 300,
            "steps": [{
                    "show_avg": "1",
                    "power": 0.5,
                    "duration": 300
                }]
        }
    ],
    "id": "84a1d5b3-471c-4218-894c-29a4aa076907",
    "intervals_id": 53218209
}
```

But in Intervals the workout has the following power targets actually defined:
```
- 10m 42%
- 10m 58%
- 10m 62%
- 10m 67%
- 10m 62%
- 10m 67%
- 3m 70%
- 2m 58%
- 5m 50%
```

Which means that between the base64encoding and decoding will get those errors:
```
42% -> 0.417 , -0.003
58% -> 0.578 , -0.002
62% -> 0.619 , -0.001
67% -> 0.669 , -0.001
62% -> 0.619 , -0.001 
67% -> 0.669 , -0.001
70% -> 0.698 , -0.002
58% -> 0.578 , -0.002 
50% -> 0.5   , 0
```

The error seems to be in the range 0 to -0.003. So this when rounded down (ceil) will result in a 0.01 / 1% lower target.

When exporting the same workouts from the Intervals.icu website with the Download Button as ZWO (Zwift) here is the output:
```
<?xml version="1.0" encoding="UTF-8" standalone="yes"?>
<workout_file>
    <author>Dimitar Marinov</author>
    <name>Kuusi Kalaa</name>
    <sportType>bike</sportType>
    <tags/>
    <workout>
        <SteadyState show_avg="1" Power="0.417" Duration="600"/>
        <SteadyState show_avg="1" Power="0.578" Duration="600"/>
        <SteadyState show_avg="1" Power="0.619" Duration="600"/>
        <SteadyState show_avg="1" Power="0.669" Duration="600"/>
        <SteadyState show_avg="1" Power="0.619" Duration="600"/>
        <SteadyState show_avg="1" Power="0.669" Duration="600"/>
        <SteadyState show_avg="1" Power="0.698" Duration="180"/>
        <SteadyState show_avg="1" Power="0.578" Duration="120"/>
        <SteadyState show_avg="1" Power="0.5" Duration="300"/>
    </workout>
</workout_file>
```

This output is the save as the decoded base64 workouts by the parser.
Therefore the 0 to 0.003 precision error is most likely happening in the Intervals.icu zwo parser-printer.

davidtinker, 2025-01-24:
The Intervals.icu ZWO file rounding issues have been fixed:

https://forum.intervals.icu/t/auuki-com-intervals-icu-integration/87105/46

dvmarinoff, 2025-01-24:
Hey, thanks for the quick response!

closed 2025-01-24 by dvmarinoff