dvmarinoff/Auuki #237: QZ Integration

cagnulein, 2025-01-27:
Hi @dvmarinoff great job with this app!

I was wondering to run your code on the QZ app https://github.com/cagnulein/qdomyos-zwift

Since QZ allows connection to all the custom bluetooth protocol, it will open Auuki to a lot of different devices!

QZ has already a web server so i was thinking to populate the metrics directly with some API.

Is something that you have already in mind? I can create a PR in case if you could address me to the best approach on your code

Let me know!

cagnulein, 2025-01-27:
qz supports basically all the wifi and bluetooth fitness devices  FTMS (but you are supported them) and NON FTMS (means custom protocols).
So what i would like to do, it's to embeed your web application in qz so the QZ users can have, if they want, your webapp directly integrated in qz without the need of open qz and your webapp on 2 different devices.

also this will solve the fact that auuki is not working on ios (QZ already can).

I guess it's very easy to do, i just need to know where i can put some code to bypass the bluetooth communication on your code, and put directly the metrics from my c++ library.

If you want i can try to do this myself, but it will be nice to cooperate on this so you can have this API ready for others project too!

dvmarinoff, 2025-01-27:
I would like to understand why the User would want to use both apps at the same time.

Auuki already supports the non-FTMS protocols and legacy device are out of scope since structured training won't make sense on them.

Another issue is that the licensing of Auuki  is absolutely incompatible with the Apple App Store.

cagnulein, 2025-01-27:
> I would like to understand why the User would want to use both apps at the same time.

because i received some request to use your app because they like your UI interface :) 

> Auuki already supports the non-FTMS protocols and legacy device are out of scope since structured training won't make sense on them.

custom protocols are handled by qz, so qz is translating ftms requests that you are doing to qz, to custom requests. so all of this is already handled by qz. 

> Another issue is that the licensing of Auuki is absolutely incompatible with the Apple App Store.

hah ok, i didn't know this. So I guess it's no possible :)

It's a pity because it would be nice to cooperate between 2 open sources project!

dvmarinoff, 2025-01-27:
This kind of architecture would write off most of the pros of this being a simple web application. Moreover I don't see the advantage of merging web user interface with mobile/desktop application. Hybrid has been a tough spot and it's why I choose to do this with pure vanilla web tech only. It's just that web interfaces work best on the web in my opinion.

dvmarinoff, 2025-01-27:
If you just want to build a similar interface your probably much better off by rebuilding it with something like Flutter (https://flutter.dev/) or Rusts' Tauri (https://v2.tauri.app/)

cagnulein, 2025-01-27:
Ok no problem ;)

closed 2025-01-27 by cagnulein