dvmarinoff/Auuki #246: Intervals.icu not syncing workouts

semaj423, 2025-02-07:
Not able to see intervals.icu workouts. I've tried disconnecting and reconnecting the sync and logging out and back on.

dvmarinoff, 2025-02-07:
Thanks for the report!

So I checked the logs and I can see some errors with Intervals.icu for at least one account. It fails to parse the response from intervals. The odd thing is that it works with others including my accounts. Will look into it.

semaj423, 2025-02-07:
Welcome! Thank you for Auuki and looking into this. :)

dvmarinoff, 2025-02-08:
So this seems to be an issue with the format of the data being send from Intervals.icu.

- Hypothesis One, There is some non-essential field missing in the data from Intervals

I did some changes that are more relaxed on the way the permitted Intervals.icu data, so if anything is missing the app will try to proceed. This case should be handled more gracefully from now on.

- Hypothesis Two, There is something odd in the Planned Events List data that is being send, but shouldn't

Like do you have anything, but workouts on your Calender in Intervals.icu for the day?

Anyway let me know if it works for you with the latest changes!

semaj423, 2025-02-08:
> So this seems to be an issue with the format of the data being send from Intervals.icu.
>
> * Hypothesis One, There is some non-essential field missing in the data from Intervals
>
> I did some changes [...]

Good points! Actually, I do have my gcal synced to intervals to show meetings I have throughout the day on intervals.

dvmarinoff, 2025-02-10:
Hey, did you had any success syncing workouts after the changes?

I am currently trying to figure out how to recreate the issue, but I can't seem to find how to add google calendar events to Intervals.icu.

semaj423, 2025-02-10:
Yes it seems to sync when there isn’t a Google calendar event associated
with it.

Have you tried connecting your personal gcal to recreate the potential
issue?

dvmarinoff, 2025-02-11:
OK managed to sync a google calendar to Intervals.icu, and indeed this is the issue. Google calendar events are being send along with workouts and are also marked as "workouts". So I added code that can handle this on my end, but will also let Interval.icu know.

Will leave here some technical notes:

Auuki is filtering for "WORKOUT" category in the request to the Intervals.icu API:

```rust
    let u = Url::parse_with_params(
        &format!("https://intervals.icu/api/v1/athlete/{intervals_athlete_id}/events"),
        &[
            ("oldest", params.oldest),
            ("newest", params.newest),
            ("category", "WORKOUT"),
            ("limit", "7"),
            ("ext", "zwo"),
        ]).map_err(|e| {
            tracing::error!("list events url parse {e}");
            Error::IntervalsUrlParseFail
        })?;
``` 

Then Deserializes the json body with EvenList and sends it to the front-end PWA:
```rust
#[derive(Debug, Deserialize, Serialize)]
pub struct Event {
    pub id: Option<i64>,
    pub start_date_local: Option<String>,
    pub category: Option<String>,
    pub name: Option<String>,
    pub indoor: Option<bool>,
    pub workout_filename: Option<String>,
    pub workout_file_base64: Option<String>,
}

type EventsList = Vec<Event>;
```

The PWA receives this:

[image]

And here is what's in the Calendar for the day:

[image]

davidtinker, 2025-02-11:
It is possible to create Intervals.icu workouts via external calendars and workouts don't have to have steps so it is ok for Intervals.icu to be returning these. You should probably filter out workouts with null workout_doc or empty workout_doc.steps.

dvmarinoff, 2025-02-11:
Yes, I see, so you can create and sync workouts from other calendars. I've updated the code to filter on the front-end too.
But just to note that events from other calendars are marked as a Workout by default, and they are not always workouts.

[image]

Anyway it's resolved as for as Auuki is concerned so closing.

@semaj423 Thanks for the report and helping solve this one!