dvmarinoff/Auuki #247: web-bluetooth crashing on failed connect()

mgibbs189, 2025-02-10:
Hey Dimitar!

Hope you've been doing well. This isn't specific to Auuki, but figured you'd also have a vested interested in seeing this Chromium bug resolved: https://issues.chromium.org/issues/40502943

You're already aware that `connect()` calls can't be aborted. Worse, `connect()` often breaks web-bluetooth entirely after a few minutes w/out a successful connection.

Basically, I'm trying to push the Chromium team to finally get this fixed, but it seems to be a glacially slow process. Perhaps you could also weigh in (they seem to care a lot about affected userbase size).

p.s. one workaround -- watching for advertisements -- requires getting users to toggle experimental flags.

Thanks,
Matt

dvmarinoff, 2025-02-11:
Hey, Matt

Sure, will join the issue. Otherwise as discussed before the re-connect story on web-bluetooth is not ideal. WatchAdvertisements is very slow, the Scan API appears to be on pause, and yes the `connect()` issue.
Given that Biketerra has events and races, this is important for Users.

The thing is that the Bluetooth stack is only OK on Android and Apple (iOS, Mac), Windows has been a train wreck since forever, and everyone is having issues there. If the Chromium team somehow manages to solve this, Web-Bluetooth will become like the preferred option there.

Otherwise I'd still recommend sticking to the platforms that already have good support for bluetooth and just develop native companion apps for Android and iOS that bridge the connection. The re-connects on iOS are so fast that most of the time I don't have a chance to even see the dropouts, I only find about them in the logs.

mgibbs189, 2025-02-11:
Thanks, Dimitar!