dvmarinoff/Auuki #251: Intervals.icu syncing workouts / timezone issue

quietvoid, 2025-02-25:
Hello.

I noticed there's an issue with planned workouts when the day is different on the server side.
It's currently 20H EST and a workout made for today (2025-02-24) does not sync.

However if I copy it into tomorrow, then it does show up.
I'm thinking there's a difference between the timezones and the code decided the workout isn't for the current day.

Maybe the local timezone should be detected?

quietvoid, 2025-02-25:
It seems `isoDate` is kinda broken. https://github.com/dvmarinoff/Auuki/blob/master/src/utils.js#L59
The offseted date isn't used at all, so it just always returns local date to UTC.

dvmarinoff, 2025-02-25:
Hey thanks for taking time to report this issue!

dvmarinoff, 2025-02-26:
Hey I added a potential fix on the `develop` branch which is deployed on dev.auuki.com, you can login there with the same creds.

dvmarinoff, 2025-02-26:
Another attempt with 03849e5

quietvoid, 2025-02-26:
That looks like it should work, I'll try it out tonight.
Thanks!

quietvoid, 2025-02-27:
It does work.

closed 2025-02-27 by quietvoid