dvmarinoff/Auuki #259: Slope targets not switching between negative and zero properly

opentree27, 2025-03-17:
Been experiencing this issue for a while now. When going from a negative slope target to a slope target of 0, the slope target does not actually change. For example if I had the following program:

[image]

The points I have circled are where it occurs. For example once the first -3 interval ends the slope target remains at -3 for the next interval which should be 0. I am using a wahoo kickr snap which does support negative slope targets so that shouldn't be an issue. I've also just observed this happening testing in my browser with no trainer connected.

dvmarinoff, 2025-03-19:
Thanks for the report will be looking into it shortly!

dvmarinoff, 2025-03-19:
So it appears that 0.0% slope targets are missing in both the computed in memory workout and the zwo file:

[image]

``` xml
<workout_file>
    <author>Auuki</author>
    <name>Slope Target Test</name>
    <category>Debug</category>
    <subcategory></subcategory>
    <description>Best workout ever!</description>
    <sporttype>bike</sporttype>
    <tags></tags>
    <workout>
        <SteadyState Duration="60" Power="0.5" />
        <SteadyState Duration="60" Power="0.7" Slope="3" />
        <SteadyState Duration="60" Power="0.4" Slope="-3" />
        <SteadyState Duration="60" Power="0.5" />
        <SteadyState Duration="60" Power="0.7" Slope="3" />
        <SteadyState Duration="60" Power="0.4" Slope="-3" />
        <SteadyState Duration="60" Power="0.5" />
        <SteadyState Duration="60" Power="0.4" Slope="-3" />
        <SteadyState Duration="60" Power="0.6" Slope="1" />
    </workout>
</workout_file>
```

But when I manually add slope 0.0% like this, the app correctly changes the target to 0.0% slope. So the error is in the zwo editor. Will try to track it further down tomorrow:

``` xml
<workout_file>
    <author>Auuki</author>
    <name>Slope Target Test</name>
    <category>Debug</category>
    <subcategory></subcategory>
    <description>Best workout ever!</description>
    <sporttype>bike</sporttype>
    <tags></tags>
    <workout>
        <SteadyState Duration="60" Power="0.5" />
        <SteadyState Duration="60" Power="0.7" Slope="3" />
        <SteadyState Duration="60" Power="0.4" Slope="-3" />
        <SteadyState Duration="60" Power="0.5" Slope="0" />
        <SteadyState Duration="60" Power="0.7" Slope="3" />
        <SteadyState Duration="60" Power="0.4" Slope="-3" />
        <SteadyState Duration="60" Power="0.5" Slope="0" />
        <SteadyState Duration="60" Power="0.4" Slope="-3" />
        <SteadyState Duration="60" Power="0.6" Slope="1" />
    </workout>
</workout_file>
```

opentree27, 2025-03-20:
Thanks for having a look!